agherzan/meta-raspberrypi #1407: Weston crashes on mouse move

kronoidz, 2025-01-27:
## Error

Weston starts, but then crashes as soon as moving the mouse with the following output:

```
root@raspberrypi5:~# XDG_RUNTIME_DIR=/run/user/1000/ weston --backend=drm
Date: 2018-03-09 UTC
[12:46:37.147] weston 14.0.1
               https://wayland.freedesktop.org
               Bug reports to: https://gitlab.freedesktop.org/wayland/weston/issues/
               Build: 14.0.1
[12:46:37.147] Command line: weston --backend=drm
[12:46:37.147] OS: Linux, 6.6.63-v8-16k, #1 SMP PREEMPT Fri Dec  6 10:10:05 UTC 2024, aarch64
[12:46:37.147] Flight recorder: enabled
[12:46:37.147] warning: XDG_RUNTIME_DIR "/run/user/1000/" is not configured
correctly.  Unix access mode must be 0700 (current mode is 0700),
and must be owned by the user UID 0 (current owner is UID 1000).
Refer to your distribution on how to get it, or
http://www.freedesktop.org/wiki/Specifications/basedir-spec
on how to implement it.
[12:46:37.147] Using config file '/etc/xdg/weston/weston.ini'
[12:46:37.147] Output repaint window is 7 ms maximum.
[12:46:37.147] WARNING: Weston was started from /dev/ttyAMA10. Messages will be dropped if not written to a file.
               Use --log for capturing Weston logs to a file.
[12:46:37.147] Loading module '/usr/lib/libweston-14/drm-backend.so'
[12:46:37.148] initializing drm backend
[12:46:37.148] Trying libseat launcher...
00:02:25.771  [c/debug/seatd/0.9.1/seatd/server.c:145] New client connected (pid: 568, uid: 0, gid: 0)
00:02:25.771  [c/debug/seatd/0.9.1/seatd/seat.c:239] Added client 1 to seat0
00:02:25.772  [c/debug/seatd/0.9.1/seatd/seat.c:563] Opened client 1 on seat0
[12:46:37.148] [c/debug/seatd/0.9.1/libseat/libseat.c:73] Seat opened with backend 'seatd'
[12:46:37.148] [c/debug/seatd/0.9.1/libseat/backend/seatd.c:212] Enabling seat
[12:46:37.148] libseat: session control granted
[12:46:37.149] using /dev/dri/card0
[12:46:37.149] DRM: supports atomic modesetting
[12:46:37.149] DRM: supports GBM modifiers
[12:46:37.149] DRM: does not support Atomic async page flip
[12:46:37.149] DRM: supports picture aspect ratio
[12:46:37.160] Loading module '/usr/lib/libweston-14/gl-renderer.so'
[12:46:37.163] Using rendering device: /dev/dri/renderD128
[12:46:37.163] EGL version: 1.4
[12:46:37.163] EGL vendor: Mesa Project
[12:46:37.163] EGL client APIs: OpenGL OpenGL_ES 
[12:46:37.164] EGL features:
               EGL Wayland extension: yes
               context priority: no
               buffer age: yes
               partial update: no
               swap buffers with damage: no
               configless context: yes
               surfaceless context: yes
               dmabuf support: modifiers
[12:46:37.164] GL version: OpenGL ES 3.1 Mesa 24.0.7
[12:46:37.164] GLSL version: OpenGL ES GLSL ES 3.10
[12:46:37.164] GL vendor: Broadcom
[12:46:37.164] GL renderer: V3D 7.1.7
[12:46:37.164] warning: Disabling render GPU timeline due to missing GL_EXT_disjoint_timer_query extension
[12:46:37.169] GL ES 3.1 - renderer features:
               read-back format: ARGB8888
               glReadPixels supports y-flip: yes
               glReadPixels supports PBO: yes
               wl_shm 10 bpc formats: yes
               wl_shm 16 bpc formats: no
               wl_shm half-float formats: no
               internal R and RG formats: yes
               OES_EGL_image_external: yes
[12:46:37.170] Using GL renderer
[12:46:37.174] event4  - SIGMACHIP Trust Keyboard: is tagged by udev as: Keyboard
[12:46:37.174] event4  - SIGMACHIP Trust Keyboard: device is a keyboard
[12:46:37.175] event6  - SIGMACHIP Trust Keyboard Consumer Control: is tagged by udev as: Keyboard
[... 100 lines cut ...]
root@raspberrypi5:~# 00:02:28.269  [c/debug/seatd/0.9.1/seatd/seat.c:158] No clients on seat0 to activate
00:02:28.320  [c/debug/seatd/0.9.1/seatd/seat.c:290] Removed client 1 from seat0
```

Relevant part of the error:

```
weston: /usr/src/debug/weston/14.0.1/libweston/backend-drm/state-propose.c:533: drm_output_find_plane_for_view: Assertion `fb' failed.
```

## Steps to reproduce

* clone *meta-raspberrypi* -- current HEAD at the time of writing: `880d36b7c2be9cf7fa7cb287d8eefe3a27d7ede9`;
* clone *poky* -- current HEAD: `59db27de565fb33f9e4326e76ebd6fa3935557b9`;
* run `source poky/oe-init-build-env rpi-build`;
* add `meta-raspberrypi` to `bblayers.conf`;
* set `MACHINE=raspberrypi5` and `LICENSE_FLAGS_ACCEPTED += "synaptics-killswitch"` in `local.conf`;
* build image: `bitbake core-image-weston`;
* flash uSD and boot Raspberry Pi 5 with mouse and keyboard connected;
* run `XDG_RUNTIME_DIR=/run/user/1000/ weston`;
* weston starts;
* move mouse;
* error.

Also tested on branch scarthgap, and no error occurs (meta-raspberry: `6df7e028a2b7b2d8cab0745dc0ed2eebc3742a17`, poky: `2240b92d62618d6125bd825647392d3bb5a2d038`).

kronoidz, 2025-01-28:
The issue appears to be specific to kernel version 6.6.63.

It's introduced by: https://github.com/agherzan/meta-raspberrypi/pull/1392, https://github.com/agherzan/meta-raspberrypi/pull/1400.

janeczkos, 2025-02-03:
Problem is caused by this line:
/usr/src/debug/weston/14.0.1/libweston/backend-drm/state-propose.c:533
There is assert(fb) which cause crash. Just remove it and it should work. Weston main branch has already change this code but there is no release version yet.

sefram, 2025-02-26:
> Problem is caused by this line:
> /usr/src/debug/weston/14.0.1/libweston/backend-drm/state-propose.c:533
> There is assert(fb) which cause crash. Just remove it and it should work. Weston main branch has already change this code but there is no release version yet.

Hello,
I try remove the assert(fb) and now it doesn't crash. BUT **on cm5,** the color is wrong, like if the configuration of framebuffer was not good...
On PI5, it works...

d99roric, 2025-10-05:
14.0.2 does not solve the assert for me. I use scarthgap as base, and up'ing weston to 14.0.2. 
Removing the assert seems to work.

kraj, 2025-11-20:
@d99roric can you try couple of things

1. Use 6.6.78 kernel, its proposed for master here - https://github.com/agherzan/meta-raspberrypi/pull/1547
2. Try using 6.12 kernel and see if issue happens with 6.12 as well.

OldManYellsAtCloud, 2025-11-21:
It looks like the culprit is [this](https://github.com/raspberrypi/linux/commit/8181e682d6f4ef209845ec24f0a1eb37764d6731) commit. They changed some 32-bit ints to 64-bit ones, and weston seems to be unhappy about it. It can be reverted at least in 6.6 cleanly - it's an rpi-specific change, not upstreamed. Currently trying to figure out if it would be easier to patch weston.

ForbiddenEra, 2026-01-16:
Weird

I was experimenting with running multi-seat stuff

I had wayfire running on an HDMI display and Weston running on a SPI display but on the same seat - when it was running like this, the mouse worked.

Then I killed the main desktop on HDMI (z2w embedded project) and moved weston over (to configure some stuff, 280x240 isnt big) and ran into this.

Not sure if that's helpful at all. I don't even see a cursor, though I did originally when both were active.