agherzan/meta-raspberrypi #1409: Freeze on Reboot (CM4, scarthgap, RAUC)

berkutta, 2025-01-29:
I'm using the meta-raspberrypi layer (scarthgap) on CM4. 

I have two root filesystems (A/B), which I update via RAUC in conjunction with rauc-hawkbit-updater.
After RAUC finishes an update cycle, it performs a reboot using the reboot shell command (post-install hook).
This works fine from an update perspective, but the CM4 often freezes during shutdown.
There is also often kernel backtrace attached. The system doesn't recover itself. It needs an power cycle. After that, it boots fine again.

The exact Kernel Backtrace varies from Kernel version to Kernel version. This is one example (The system hangs forever after this output via Serial, this is the last Serial output, Power cycle needed):
```
[  177.933557] systemd-shutdown[1]: All filesystems unmounted.
[  177.939284] systemd-shutdown[1]: Deactivating swaps.
[  177.944496] systemd-shutdown[1]: All swaps deactivated.
[  177.949923] systemd-shutdown[1]: Detaching loop devices.
[  177.960674] systemd-shutdown[1]: All loop devices detached.
[  177.966394] systemd-shutdown[1]: Stopping MD devices.
[  177.971775] systemd-shutdown[1]: All MD devices stopped.
[  177.977201] systemd-shutdown[1]: Detaching DM devices.
[  177.982592] systemd-shutdown[1]: All DM devices detached.
[  177.988096] systemd-shutdown[1]: All filesystems, swaps, loop devices, MD devices and DM devices detached.
[  178.001703] systemd-shutdown[1]: Syncing filesystems and block devices.
[  178.008679] systemd-shutdown[1]: Rebooting with argument '0 tryboot'.
[  218.845608] mmc0: card 0001 removed
[  218.954746] EXT4-fs (mmcblk0p2): shut down requested (2)
[  218.960210] Aborting journal on device mmcblk0p2-8.
[  363.485953] INFO: task systemd-shutdow:1 blocked for more than 120 seconds.
[  363.493077]       Tainted: G         C         6.6.30-v8 #1
[  363.498752] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  363.506717] task:systemd-shutdow state:D stack:0     pid:1     ppid:0      flags:0x00000004
[  363.515221] Call trace:
[  363.517704]  __switch_to+0xe0/0x160
[  363.521264]  __schedule+0x37c/0xd68
[  363.524816]  schedule+0x64/0x108
[  363.528102]  schedule_timeout+0x1a8/0x1c0
[  363.532186]  wait_for_completion+0xc8/0x170
[  363.536444]  __flush_work.isra.0+0x190/0x2e8
[  363.540794]  __cancel_work_timer+0x124/0x1b0
[  363.545139]  cancel_delayed_work_sync+0x1c/0x30
[  363.549753]  __mmc_stop_host+0x44/0x60
[  363.553570]  mmc_host_classdev_shutdown+0x1c/0x38
[  363.558360]  device_shutdown+0x138/0x268
[  363.562357]  kernel_restart+0x48/0xd0
[  363.566087]  __do_sys_reboot+0x1b4/0x258
[  363.570083]  __arm64_sys_reboot+0x2c/0x40
[  363.574166]  invoke_syscall+0x50/0x128
[  363.577986]  el0_svc_common.constprop.0+0x48/0xf0
[  363.582775]  do_el0_svc+0x24/0x38
[  363.586152]  el0_svc+0x40/0xe8
[  363.589258]  el0t_64_sync_handler+0x120/0x130
[  363.593694]  el0t_64_sync+0x190/0x198
[  363.597437] INFO: task kworker/2:4:68 blocked for more than 120 seconds.
[  363.604255]       Tainted: G         C         6.6.30-v8 #1
[  363.609923] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  363.617891] task:kworker/2:4     state:D stack:0     pid:68    ppid:2      flags:0x00000008
[  363.626395] Workqueue: events_freezable mmc_rescan
[  363.631275] Call trace:
[  363.633756]  __switch_to+0xe0/0x160
[  363.637307]  __schedule+0x37c/0xd68
[  363.640856]  schedule+0x64/0x108
[  363.644140]  io_schedule+0x44/0x68
[  363.647602]  bit_wait_io+0x20/0x80
[  363.651065]  __wait_on_bit+0x58/0x180
[  363.654789]  out_of_line_wait_on_bit+0x90/0xc8
[  363.659312]  __wait_on_buffer+0x34/0x48
[  363.663218]  jbd2_write_superblock+0x13c/0x2f0
[  363.667743]  jbd2_journal_update_sb_errno+0x5c/0x98
[  363.672708]  jbd2_journal_abort+0x88/0x100
[  363.676878]  ext4_force_shutdown+0x1e0/0x1f8
[  363.681223]  ext4_shutdown+0x1c/0x30
[  363.685788]  fs_bdev_mark_dead+0x68/0xb0
[  363.690701]  bdev_mark_dead+0x4c/0xe0
[  363.695352]  blk_report_disk_dead+0x88/0xf8
[  363.700537]  del_gendisk+0x320/0x348
[  363.705089]  mmc_blk_remove+0xa4/0x248
[  363.709800]  mmc_bus_remove+0x28/0x40
[  363.714421]  device_remove+0x54/0x90
[  363.718957]  device_release_driver_internal+0x1d4/0x238
[  363.725162]  device_release_driver+0x20/0x38
[  363.730389]  bus_remove_device+0xd4/0x120
[  363.735330]  device_del+0x164/0x430
[  363.739732]  mmc_remove_card+0x60/0xe0
[  363.744398]  mmc_detect+0x5c/0xa0
[  363.748615]  mmc_rescan+0x94/0x328
[  363.752914]  process_one_work+0x148/0x3b8
[  363.757824]  worker_thread+0x32c/0x450
[  363.762467]  kthread+0x118/0x128
[  363.766573]  ret_from_fork+0x10/0x20
```

Sometimes it also just hangs without Backtrace. In this case the Kernel even still printed some messages (The system hangs forever after this output via Serial, this is the last Serial output, Power cycle needed):
```
[  116.566952] systemd-shutdown[1]: All filesystems unmounted.
[  116.572818] systemd-shutdown[1]: Deactivating swaps.
[  116.578075] systemd-shutdown[1]: All swaps deactivated.
[  116.583413] systemd-shutdown[1]: Detaching loop devices.
[  116.597291] systemd-shutdown[1]: All loop devices detached.
[  116.603015] systemd-shutdown[1]: Stopping MD devices.
[  116.608583] systemd-shutdown[1]: All MD devices stopped.
[  116.614019] systemd-shutdown[1]: Detaching DM devices.
[  116.619598] systemd-shutdown[1]: All DM devices detached.
[  116.625137] systemd-shutdown[1]: All filesystems, swaps, loop devices, MD devices and DM devices detached.
[  116.638948] systemd-shutdown[1]: Syncing filesystems and block devices.
[  116.646282] systemd-shutdown[1]: Rebooting with argument '0 tryboot'.
[21626.827418] IPv4: martian source 255.255.255.255 from 192.168.0.248, on dev eth0
[21626.834971] ll header: 00000000: ff ff ff ff ff ff d8 eb 46 86 fd 39 08 00
[21628.434569] IPv4: martian source 255.255.255.255 from 192.168.0.56, on dev eth0
[21628.442006] ll header: 00000000: ff ff ff ff ff ff b4 8a 0a b1 6c 90 08 00
[...]
[23703.944066] IPv4: martian source 255.255.255.255 from 192.168.0.56, on dev eth0
[23703.951501] ll header: 00000000: ff ff ff ff ff ff b4 8a 0a b1 6c 90 08 00
[23708.944633] IPv4: martian source 255.255.255.255 from 192.168.0.56, on dev eth0
[23708.952069] ll header: 00000000: ff ff ff ff ff ff b4 8a 0a b1 6c 90 08 00
[23712.222015] bcmgenet fd580000.ethernet eth0: Link is Down
[23715.294285] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
```

Anyone ever encountered such an issue while using CM4/meta-raspberrypi?

mlilien, 2025-02-11:
i encounter this issue with a rpi4 + meta-raspberrypi + scarthgap.
i use swupdate and not rauc, but imho that is irrelevant. 

workaround: set kernel parameter via
```
CMDLINE:append:pn-rpi-cmdline = " bcm2835-wdt.nowayout=1"
```

berkutta, 2025-02-11:
@mlilien Nice to hear from someone encountering the same (or similar) problem!
Also thanks for the workaround, nice idea to use that WDT parameter.

Have you found a good way to reproduce this?
My current way is to update the CM4 with RAUC in a loop. This way it shows the error within 1-5 days.

Do you have a DSI Display connected to the RPI4?

mlilien, 2025-02-12:
> Have you found a good way to reproduce this?
Unfortunately no. 

> Do you have a DSI Display connected to the RPI4?
No, the system is headless.

OldManYellsAtCloud, 2025-02-23:
Random guess: do you have this change in your kernel tree?
https://github.com/raspberrypi/linux/commit/07f395d6197cd2f61367a1dc6f24252b5c061014

If I read the devicetree (and the patch) correctly, broken-cd is set on the rpi4 sd-card, so it is being polled - and during shutdown the card could be detected again in some cases, after it had been disabled.

tobiasjakobi-lr, 2025-04-16:
You can also instruct systemd to configure the HW watchdog to a shorter time during reboot, see [systemd-system.conf](https://www.freedesktop.org/software/systemd/man/latest/systemd-system.conf.html#RuntimeWatchdogSec=) -- the default timeout is 10 minutes.

Of course that's only the workaround version.

What puzzles me is the order of outputs in the first log. At the point where `Rebooting with argument` is printed, all filesystems should be either fully unmounted or at least be remounted r/o. This doesn't seem to be the case here, since mmcblk0p2 is still active r/w and the card removal triggers a write of the journal.

I would try to find out what `mmcblk0p2` is (rootfs?which one?) and who mounted it in the first place. If mounting was done through normal systemd/fstab mechanism then the fs should be unmounted when `All filesystems unmounted` is printed.

tobiasjakobi-lr, 2025-04-16:
@berkutta systemd never disables the HW watchdog (if you let it use it in the first place). During normal operation it honors the `RuntimeWatchdogSec=` setting, and then, once it enters the reboot phase, it reprograms the watchdog to use the `RebootWatchdogSec=` setting. There are some watchdogs where you can't reprogram the timeout. Hence you stay on the previous timeout.

dwalkes, 2025-11-19:
Confirmed `bcm2835-wdt.nowayout=1` workaround works on CM5 in my scenario as well.

kraj, 2025-11-19:
> I suspect the hang is happening in the mmc stack somewhere and using `nowayout` means the watchdog completes the reboot instead of being hung forever in the kernel waiting for some mmc process to complete which isn't actually completing. Since I have a way to reproduce I might experiment with later kernel versions to see if I can find a difference.

If using SD card it might be worth trying a different make as well

dwalkes, 2025-11-19:
Thanks, not using SDCard though, using CM5 with internal emmc.

berkutta, 2025-11-19:
It would be very interesting if someone is able to reproduce this issue on something like Raspberry Pi OS. This way we could try to convince Raspberry Pi to look at this.

Unfortunately, I wasn't able to ever reproduce this on Raspberry Pi OS. At the same time I also don't know what the difference is (or if my testing was flawed).

(I'm kind of in the same boat that eMMC might be the culprit)

dspiller, 2025-12-04:
I have a very similar issue to this with my CM4 on kernel 6.6.63 in Scarthgap.  My critical problem reduces significantly if we set nowayout=1, but I've reproduced the issue even with nowayout=1, which is hard to understand.  Watchdog is set to 15s.

It took around 100 attempts to reproduce it with nowayout turned on, but without it, I was consistently reproducing it within 10-20 tests.

```[   21.262896] watchdog: watchdog0: nowayout prevents watchdog being stopped!
[   21.262896] watchdog: watchdog0: nowayout prevents watchdog being stopped!
[   21.269877] systemd-shutdown[1]: Failed to disable hardware watchdog, ignoring: Device or resource busy
[   21.279415] watchdog: watchdog0: nowayout prevents watchdog being stopped!
[   21.286388] watchdog: watchdog0: watchdog did not stop!
[   21.292075] systemd-shutdown[1]: Unable to finalize remaining file systems, loop devices, ignoring.
[   21.301658] systemd-shutdown[1]: Syncing filesystems and block devices.
[   21.308508] systemd-shutdown[1]: Rebooting.
[   62.207498] mmc0: card 0001 removed
[   62.247983] EXT4-fs (mmcblk0p2): shut down requested (2)
[   62.253388] Aborting journal on device mmcblk0p2-8.
[  242.689826] INFO: task systemd-shutdow:1 blocked for more than 120 seconds.
[  242.696910]       Tainted: G         C         6.6.63-v8 #1
[  242.702562] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  242.710502] task:systemd-shutdow state:D stack:0     pid:1     ppid:0      flags:0x0000000c
[  242.718980] Call trace:
[  242.721453]  __switch_to+0xd8/0x140
[  242.724994]  __schedule+0x37c/0xd68
[  242.728530]  schedule+0x64/0x108
[  242.731801]  schedule_timeout+0x1a8/0x1c0
[  242.735867]  wait_for_completion+0xc8/0x170
[  242.740107]  __flush_work.isra.0+0x184/0x2d0
[  242.744439]  __cancel_work_timer+0x124/0x1b0
[  242.748769]  cancel_delayed_work_sync+0x1c/0x30
[  242.753363]  __mmc_stop_host+0x44/0x60
[  242.757166]  mmc_host_classdev_shutdown+0x1c/0x38
[  242.761937]  device_shutdown+0x138/0x268
[  242.765917]  kernel_restart+0x48/0xd0
[  242.769628]  __do_sys_reboot+0x178/0x270
[  242.773606]  __arm64_sys_reboot+0x2c/0x40
[  242.777672]  invoke_syscall+0x50/0x128
[  242.781476]  el0_svc_common.constprop.0+0x48/0xf0
[  242.786253]  do_el0_svc+0x24/0x38
[  242.789611]  el0_svc+0x38/0xc8
[  242.793215]  el0t_64_sync_handler+0x120/0x130
[  242.797636]  el0t_64_sync+0x190/0x198
[  242.801356] INFO: task kworker/1:0:23 blocked for more than 120 seconds.
[  242.808151]       Tainted: G         C         6.6.63-v8 #1
[  242.813800] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  242.821743] task:kworker/1:0     state:D stack:0     pid:23    ppid:2      flags:0x00000008
[  242.830219] Workqueue: events_freezable mmc_rescan
[  242.835079] Call trace:
[  242.837551]  __switch_to+0xd8/0x140
[  242.841086]  __schedule+0x37c/0xd68
[  242.844622]  schedule+0x64/0x108
[  242.847892]  io_schedule+0x44/0x68
[  242.851339]  bit_wait_io+0x20/0x80
[  242.854787]  __wait_on_bit+0x58/0x180
[  242.858498]  out_of_line_wait_on_bit+0x90/0xc8
[  242.863004]  __wait_on_buffer+0x34/0x48
[  242.866894]  jbd2_write_superblock+0x13c/0x2f0
[  242.871401]  jbd2_journal_update_sb_errno+0x5c/0x98
[  242.876347]  jbd2_journal_abort+0x88/0x100
[  242.880501]  ext4_force_shutdown+0x1e0/0x1f8
[  242.884834]  ext4_shutdown+0x1c/0x30
[  242.888458]  fs_bdev_mark_dead+0x68/0xb0
[  242.892436]  bdev_mark_dead+0x4c/0xe0
[  242.896152]  blk_report_disk_dead+0x88/0xf8
[  242.900394]  del_gendisk+0x320/0x348
[  242.904019]  mmc_blk_remove+0xa4/0x248
[  242.907822]  mmc_bus_remove+0x28/0x40
[  242.911534]  device_remove+0x54/0x90
[  242.915159]  device_release_driver_internal+0x1d4/0x238
[  242.920459]  device_release_driver+0x20/0x38
[  242.924790]  bus_remove_device+0xd4/0x120
[  242.928856]  device_del+0x164/0x430
[  242.932392]  mmc_remove_card+0x60/0xe0
[  242.936192]  mmc_detect+0x5c/0xa0
[  242.939552]  mmc_rescan+0x94/0x330
[  242.943001]  process_one_work+0x148/0x3b8
[  242.947067]  worker_thread+0x32c/0x450
[  242.950867]  kthread+0x118/0x128
[  242.954139]  ret_from_fork+0x10/0x20 
```

We are also performing a kexec during the boot process without the watchdog enabled, and a delay will happen the kexec process some of the time, causing a ~40 second delay before it succeeds.  

It feels like a shutdown race that is somehow preventing a reboot, but I can't explain why the watchdog does not resolve the reboot problem 100%.

alitofresko, 2025-12-29:
I ran into an apparently identical reboot hang issue on Raspberry Pi (Yocto / Poky, systemd, swupdate, eMMC/MMC rootfs).

After some investigation, I believe the root cause has already been fixed upstream in the Raspberry Pi kernel with this commit:

Fix commit (merged):
https://github.com/raspberrypi/linux/commit/d562b457e1c362f60d5c807a3a0775a9204a66a0

(Dec 14, 2024)

However, the meta-raspberry layer for scarthgap currently pins linux-raspberrypi to an earlier commit on the 6.6 branch:

Currently used commit:
https://github.com/raspberrypi/linux/commit/e442e5c1ab6bff5b5460b4fc949beb72aaf77970

(Dec 6, 2024)

So although meta-raspberry is on kernel 6.6, it does not yet include the above fix.

From my testing and log analysis, the symptoms (reboot hanging during shutdown, MMC/eMMC reporting card removal, systemd-shutdown blocked in mmc_stop_host()) match exactly what is addressed by the Dec 14 commit.

At the moment, users affected by this issue have a few options:
- Backport/apply the above patch in their own Yocto layer
- Force alignment to a more recent commit of the linux-raspberrypi 6.6 branch
- Move directly to linux-raspberrypi 6.12, which is now the default upstream branch and already includes the fix

I’m sharing this in case it helps others hitting the same reboot hang, and to document that the issue appears to be resolved upstream but not yet picked up by the current meta-raspberry scarthgap kernel revision.

<details>
<summary>Logs when reboot works fine</summary>
```
[  OK  ] Reached target System Shutdown.
[  OK  ] Reached target Late Shutdown Services.
[  OK  ] Finished System Reboot.
[  OK  ] Reached target System Reboot.
[   32.428532] watchdog: watchdog0: watchdog did not stop!
[   32.443130] systemd-shutdown[1]: Using hardware watchdog 'Broadcom BCM2835 Watchdog timer', version 0, device /dev/watchdog0
[   32.454901] systemd-shutdown[1]: Watchdog running with a hardware timeout of 1min.
[   32.485593] systemd-shutdown[1]: Syncing filesystems and block devices.
[   32.492551] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   32.512935] systemd-journald[137]: Received SIGTERM from PID 1 (systemd-shutdow).
[   32.522465] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[   32.538540] systemd-shutdown[1]: Unmounting file systems.
[   32.544695] systemd-shutdown[1]: All filesystems unmounted.
[   32.550362] systemd-shutdown[1]: Deactivating swaps.
[   32.555736] systemd-shutdown[1]: All swaps deactivated.
[   32.561041] systemd-shutdown[1]: Detaching loop devices.
[   32.569731] systemd-shutdown[1]: All loop devices detached.
[   32.575425] systemd-shutdown[1]: Stopping MD devices.
[   32.580724] systemd-shutdown[1]: All MD devices stopped.
[   32.586139] systemd-shutdown[1]: Detaching DM devices.
[   32.591495] systemd-shutdown[1]: All DM devices detached.
[   32.597141] systemd-shutdown[1]: All filesystems, swaps, loop devices, MD devices and DM devices detached.
[   32.607925] systemd-shutdown[1]: Syncing filesystems and block devices.
[   32.614733] systemd-shutdown[1]: Rebooting.
[   32.760473] kvm: exiting hardware virtualization
[   32.765185] reboot: Restarting system
```
</details>

<details>
<summary>Logs when things go bad</summary>
```
[  OK  ] Reached target System Shutdown.
[  OK  ] Reached target Late Shutdown Services.
[  OK  ] Finished System Reboot.
[  OK  ] Reached target System Reboot.
[   32.352300] watchdog: watchdog0: watchdog did not stop!
[   32.367066] systemd-shutdown[1]: Using hardware watchdog 'Broadcom BCM2835 Watchdog timer', version 0, device /dev/watchdog0
[   32.378490] systemd-shutdown[1]: Watchdog running with a hardware timeout of 1min.
[   32.409064] systemd-shutdown[1]: Syncing filesystems and block devices.
[... 308 lines cut ...]
[  726.170538]  jbd2_journal_abort+0x88/0x100
[  726.174706]  ext4_force_shutdown+0x1e0/0x1f8
[  726.179055]  ext4_shutdown+0x1c/0x30
[  726.182698]  fs_bdev_mark_dead+0x68/0xb0
[  726.186695]  bdev_mark_dead+0x4c/0xe0
[  726.190425]  blk_report_disk_dead+0x88/0xf8
[  726.194687]  del_gendisk+0x320/0x348
[  726.198331]  mmc_blk_remove+0xa4/0x248
[  726.202150]  mmc_bus_remove+0x28/0x40
[  726.205881]  device_remove+0x54/0x90
[  726.209518]  device_release_driver_internal+0x1d4/0x238
[  726.214832]  device_release_driver+0x20/0x38
[  726.219175]  bus_remove_device+0xd4/0x120
[  726.223252]  device_del+0x164/0x430
[  726.226807]  mmc_remove_card+0x60/0xe0
[  726.230620]  mmc_detect+0x5c/0xa0
[  726.233996]  mmc_rescan+0x94/0x330
[  726.237469]  process_one_work+0x148/0x3b8
[  726.241551]  worker_thread+0x324/0x448
[  726.245366]  kthread+0x118/0x128
[  726.248649]  ret_from_fork+0x10/0x20
[  726.252290] Future hung task reports are suppressed, see sysctl kernel.hung_task_warnings

```
</details>

berkutta, 2025-12-29:
> After some investigation, I believe the root cause has already been fixed upstream in the Raspberry Pi kernel with this commit:

@alitofresko Were you able to verify that the problem goes away with this patch present?

This patch is also available in the 6.6 branch: https://github.com/raspberrypi/linux/commit/07f395d6197cd2f61367a1dc6f24252b5c061014

IIRC I wasn't able to verify this fixes the problem 100% even with this patch applied. But my memory can be wrong!

This is also the reason why I never continued with https://github.com/agherzan/meta-raspberrypi/issues/1404

dwalkes, 2025-12-31:
After testing today with https://github.com/Trellis-Logic/meta-raspberrypi/tree/scarthgap%2Blinux-6.6.78 it definitely changes the problem but does not completely resolve it, at least for me.  When I backport https://github.com/agherzan/meta-raspberrypi/commit/558f3ac5e6fa05aaa52a65d2e6525aeda8afc022 to scarthgap and get
```
uname -r
6.6.78-v8-16k
```
Then remove `bcm2835-wdt.nowayout=1` from `/boot/cmdline.txt` I see that I no longer get the "Aborting journal" on reboot, the boot messages look like this on 6.6.63
```
[   33.550441] systemd-shutdown[1]: Rebooting.
[   55.114636] mmc1: Failed to initialize a non-removable card
[   74.823958] mmc0: card 0001 removed
[   74.877972] EXT4-fs (mmcblk0p2): shut down requested (2)
[   74.885312] Aborting journal on device mmcblk0p2-8.
```
with an indefinite hang at "Aborting journal"
to a list like this on 6.6.78:
```
[   54.415820] systemd-shutdown[1]: Rebooting.
[   54.874752] mmc1: Failed to initialize a non-removable card
[   54.874887] macb 1f00100000.ethernet eth0: Link is Down
[   54.886080] macb 1f00100000.ethernet: gem-ptp-timer ptp clock unregistered.
[   54.894336] kvm: exiting hardware virtualization
[   54.898972] reboot: Restarting system
```
and the system restarts successfully.  However, on the very next (warm) boot I now get:
```
[    3.125190] mmc0: SDHCI controller on 1000fff000.mmc [1000fff000.mmc] using ADMA 64-bit
[    3.194937] usb 3-1: New USB device found, idVendor=1a86, idProduct=8091, bcdDevice=13.20
[    3.203150] usb 3-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[    3.210315] usb 3-1: Product: USB HUB
[    3.254225] hub 3-1:1.0: USB hub found
[    3.258188] hub 3-1:1.0: 4 ports detected
[    3.295733] mmc1: SDHCI controller on 1001100000.mmc [1001100000.mmc] using ADMA 64-bit
[    3.319789] clk: Disabling unused clocks
[    3.331687] ALSA device list:
[    3.334673]   #0: vc4-hdmi-0
[    3.337559]   #1: vc4-hdmi-1
[    3.340484] uart-pl011 107d001000.serial: no DMA platform data
[    3.346386] Waiting for root device /dev/mmcblk0p3...
```
and the system hangs for roughly 2 minutes here waiting for the root device.  It doesn't happen every boot, happens on approximately the same frequency I saw hangs during shutdown previously.

After 2 minutes I get:
```
[    3.925504] mmc1: Failed to initialize a non-removable card
[  136.118198] mmc0: new HS400 Enhanced strobe MMC card at address 0001
[  136.130409] mmcblk0: mmc0:0001 AJTD4R 14.6 GiB
[  136.136430]  mmcblk0: p1 p2 p3 p4 < p5 p6 p7 >
[  136.141077] mmcblk0: mmc0:0001 AJTD4R 14.6 GiB
[  136.145723] mmcblk0boot0: mmc0:0001 AJTD4R 4.00 MiB
[  136.151379] mmcblk0boot1: mmc0:0001 AJTD4R 4.00 MiB
[  136.156840] mmcblk0rpmb: mmc0:0001 AJTD4R 4.00 MiB, chardev (242:0)
[  136.173275] EXT4-fs (mmcblk0p3): orphan cleanup on readonly fs
[  136.179248] EXT4-fs (mmcblk0p3): mounted filesystem 0eb8d7f8-567f-4205-954e-5a88a896a1d8 ro with ordered data mode. Quota mode: none.
```

So it seems like at least in my case the problem moved from shutdown to startup when I upgraded from `6.6.63` to `6.6.78`.  It's perhaps slightly better since it seems to recover in 2 minutes without the watchdog, but I'm wondering if this lasts longer for some reason it will mean I won't have the watchdog to save me like I did with the reboot case.

If I'm the only one seeing this issue with 6.6.78 I think I'm going to switch hardware.  I'm currently using a custom carrier board, will try to reproduce with a CM5 dev board.

alitofresko, 2026-01-07:
> So it seems like at least in my case the problem moved from shutdown to startup when I upgraded from `6.6.63` to `6.6.78`. It's perhaps slightly better since it seems to recover in 2 minutes without [...]

Hi @dwalkes, this issue was related to the “Freeze on Reboot” problem and, as of today, after 12k successful reboots, I believe it is resolved.

Regarding your ongoing issue, I think it’s a different matter. Have you tried using a clean rootfs, or is your image the same as before with only the kernel upgraded?

If you’d like help investigating the issue you’re currently facing, I’d suggest opening a new issue. If you can also provide a way to reproduce the same conditions (for example by sharing your image), I can try it on a CM4IO and see what comes up.