agherzan/meta-raspberrypi #1411: CM5 stuck on Waiting for root device /dev/mmcblk0p2...

JeanPhilippeLebel, 2025-01-31:
Hello,

I've just upgraded every to the latest meta-raspberrypi in order to test a system that was running using a CM4 but is now upgraded to a CM5

Going for other issues, I was able to enable the UART console but the boot process stucks on 

[    3.524080] Waiting for root device /dev/mmcblk0p2...

MACHINE is set to raspberrypi5

Any idea what can cause that issue?

kraj, 2025-01-31:
@JeanPhilippeLebel it seems MMC or parition number might be skewed. Can you try finding this out. Usually SD card boot helps with such issues.

JeanPhilippeLebel, 2025-02-03:
Well, I'm booting from the internal eMMC, using core-image-minimal, so I wouldn't expect the partition scheme to be different between the cm4 and cm5 ?

Thanks,

NumberOneGit, 2025-02-07:
I haven’t tried meta-raspberry yet but I am experiencing the same problem with buildroot projects on the CM5.  I will be trying to install meta-raspberry on my CM5 so will let you know if I have the same experience.

dferrarigrowtech, 2025-02-10:
I had the same issue, but I fixed it by including some `.dtb`. Here's the patch:

```diff
diff --git a/conf/machine/raspberrypi5.conf b/conf/machine/raspberrypi5.conf
index 03f15c3..1aac825 100644
--- a/conf/machine/raspberrypi5.conf
+++ b/conf/machine/raspberrypi5.conf
@@ -15,8 +15,14 @@ MACHINE_EXTRA_RRECOMMENDS += "\
 
 RPI_KERNEL_DEVICETREE = " \
     broadcom/bcm2712-rpi-5-b.dtb \
+    broadcom/bcm2712-rpi-cm5-cm5io.dtb \
+    broadcom/bcm2712-rpi-cm5-cm4io.dtb \
+    broadcom/bcm2712-rpi-cm5l-cm5io.dtb \
+    broadcom/bcm2712-rpi-cm5l-cm4io.dtb \
 "
 
+RPI_KERNEL_DEVICETREE_OVERLAYS:append = " overlays/bcm2712d0.dtbo"
+
 SDIMG_KERNELIMAGE ?= "kernel_2712.img"
 SERIAL_CONSOLES ?= "115200;ttyAMA10"
 
diff --git a/recipes-kernel/linux/linux-raspberrypi_6.6.bb b/recipes-kernel/linux/linux-raspberrypi_6.6.bb
index b4d9953..7aaf3bc 100644
--- a/recipes-kernel/linux/linux-raspberrypi_6.6.bb
+++ b/recipes-kernel/linux/linux-raspberrypi_6.6.bb
@@ -1,8 +1,8 @@
-LINUX_VERSION ?= "6.6.22"
+LINUX_VERSION ?= "6.6.74"
 LINUX_RPI_BRANCH ?= "rpi-6.6.y"
 LINUX_RPI_KMETA_BRANCH ?= "yocto-6.6"
 
-SRCREV_machine = "c04af98514c26014a4f29ec87b3ece95626059bd"
+SRCREV_machine = "a18d9ced4965462cb7b3b4252ada440395105308"
 SRCREV_meta = "6a24861d6504575a4a9f92366285332d47c7e111"
 
 KMETA = "kernel-meta"
```

For context I am on scarthgap

JeanPhilippeLebel, 2025-02-10:
Ho wow, this fixes my issue!

Can we turn this into a patch to be added to meta-raspberrypi?

JeanPhilippeLebel, 2025-02-10:
@dferrarigrowtech Makes sense.

On a, maybe, related note. I need to support both the cm4 and cm5 so I've set "MACHINE = 'raspberrypi-armv8'"

When doing so, the cm5 hangs on boot, whereas the cm4 doesn't boot at all.

dferrarigrowtech, 2025-03-22:
Almost there, at the moment most of the patches has been submitted upstream. If you are on the latest scarthgap commit (c153c694bd7a08de474470d7c1c7fd550117b162) and you want to boot a CM5 now the patch is much smaller:

```patch
diff --git a/conf/machine/raspberrypi5.conf b/conf/machine/raspberrypi5.conf
index f2f4087..1aac825 100644
--- a/conf/machine/raspberrypi5.conf
+++ b/conf/machine/raspberrypi5.conf
@@ -15,6 +15,10 @@ MACHINE_EXTRA_RRECOMMENDS += "\
 
 RPI_KERNEL_DEVICETREE = " \
     broadcom/bcm2712-rpi-5-b.dtb \
+    broadcom/bcm2712-rpi-cm5-cm5io.dtb \
+    broadcom/bcm2712-rpi-cm5-cm4io.dtb \
+    broadcom/bcm2712-rpi-cm5l-cm5io.dtb \
+    broadcom/bcm2712-rpi-cm5l-cm4io.dtb \
 "
 
 RPI_KERNEL_DEVICETREE_OVERLAYS:append = " overlays/bcm2712d0.dtbo"
```

kraj, 2025-03-22:
@dferrarigrowtech is this delta proposed in some PR ? if not please create one.

dferrarigrowtech, 2025-03-22:
Done: #1439

poett1, 2026-04-17:
I'm facing the same issue with CM5 (with eMMC) and U-Boot enabled on latest scarthgap and master branches. First boot after flashing is always fine, after that it happens 9 out of 10 times. @kraj any ideas?

ephraim, 2026-10-06:
Happens to me from time to time too. I would say 50% of a boot hangs. Has anyone found a fix for this? 

Using Scarthgap [6ca1f750](https://github.com/agherzan/meta-raspberrypi/commit/6ca1f75017cc5d5acdb8bb05634c4bc01fa049fd)

[    3.453285] Waiting for root device /dev/mmcblk0p2...
[  134.213546] mmc0: Command Queue Engine enabled, 16 tags

ephraim, 2026-10-06:
Just recognized, that due to the use of rauc for updates, I have two "root=" entries in my cmdline. The first is always "root=/dev/mmcblk0p2" which comes from the cmdline.txt in /boot. The second comes from the rauc integration into u-boot which is the correct root device. 
In my case it was "root=/dev/mmcblk0p3". 
Perhaps this is a problem.

I removed (only) the "root=....." entry from /boot/cmdline.txt and at least the very next reboot did work correctly.

Hope that is the fix and perhaps someone else can test it two!

Cheers,
Thilo

dferrarigrowtech, 2026-10-06:
I haven't had any problems since #1439

closed 2026-10-14 by agherzan

ephraim, 2026-10-21:
Sorry but for me this bug still exists.
Building yocto image scarthgap latest with linux-raspberrypi 6.12 and the given "solution" is definitley there, but not working.

Has someone another idea, what the problem is?
I'm using a Raspberrypi CM5 with eMMC.

ephraim, 2026-11-09:
Hi again! Still working on this one. Has anyone an idea, how I can doublecheck that, that which the dtbs should fix, is fixed?

AFAI can see, u-boot is pulling the correct dtb in and to bcm2712d0.dtb is added via dtoverlay in config.txt
So it should work, but doesn't :( ...

poett1, 2026-11-09:
@ephraim you are on scarthgap, right? Which u-boot version does your build use?

ephraim, 2026-11-09:
@poett1 Yes! I'm on scarthgap. I use the meta-lts-mixins for u-boot and there the branch scarthgap/u-boot-2025.04.

Is it related to u-boot not reseting the sdhci correctly?

poett1, 2026-11-09:
@ephraim yes, that's exactly it. It's a U-Boot bug, not a meta-raspberrypi or DT one, and it seems like it only shows up when U-Boot is in the boot chain on a CM5 with eMMC.

`sdhci_brcmstb_init_2712()` checks `host->mmc->host_caps`, which is still zero at probe time, so U-Boot always treats the eMMC as a removable SD slot and sets `SDIO_CFG_SD_PIN_SEL` to SD mode. That register survives into Linux. The first power-up then doesn't latch, and the kernel runs through the SDIO and SD probe phases with 10 s timeouts before the eMMC attaches. That's the ~134 s stall at "Waiting for root device".

**Fix:** U-Boot commit [8de24e226d74 "mmc: bcmstb: Fix non-removable check in bcm2712 init"](https://git.u-boot-project.org/u-boot/u-boot/-/commit/8de24e226d745c59972b1898df5cceacf0a7ad47), in v2026.07 and later. Every U-Boot from v2024.04 to v2026.04 has the bug.

- **master:** fixed already; oe-core's U-Boot 2026.10 includes the commit.
- **wrynose:** backported in 2e70eb7 (`u-boot_2026.01: Backport 8de24e226d7`).
 - **scarthgap:** the default u-boot is **2024.01**, where this issue does not exist yet.
- **scarthgap + meta-lts-mixins `scarthgap/u-boot` (2025.04):** not fixed yet. I've sent the backport to the yocto-patches list ("[meta-lts-mixins][scarthgap/u-boot][PATCH] u-boot: backport fix for the bcm2712 non-removable check"), so we don't need to maintain a patch on this repo. Until it's merged, you can carry it in your own layer.

Add `recipes-bsp/u-boot/u-boot_%.bbappend`:

```
FILESEXTRAPATHS:prepend:rpi := "${THISDIR}/files:"
SRC_URI:append:rpi = " file://8de24e226d7.patch"
```

and get the patch next to it:

```
curl -L -o recipes-bsp/u-boot/files/8de24e226d7.patch \
    https://github.com/u-boot/u-boot/commit/8de24e226d745c59972b1898df5cceacf0a7ad47.patch
```

It should apply to 2025.04 without fuzz. Drop the bbappend once the mixin carries the fix, or the patch gets applied twice.

On my CM5 this took me from 6 of 9 boots stalling at 134 s to 0 of 19.

ephraim, 2026-11-09:
@poett1 Thank you for this awesome help! I believed u-boot is the problem but all googling did not help! 
My system boots now wonderful smooth!

Now, your answer is right there where the "Waiting for root device" google leads you. That will definitely help others to, I believe.