agherzan/meta-raspberrypi #1422: openBMC on raspberry Pi 4

iankao0000, 2025-02-20:
Hello everyone,
I had a problem with trying to build openbmc on raspberrypi 4 too ,

TEMPLATECONF=meta-evb/meta-evb-raspberrypi/conf/templates/default . openbmc-env

bitbake obmc-phosphor-image

WARNING: linux-raspberrypi-1_6.6.63+git-r0 do_fetch: Failed to fetch URL git://github.com/raspberrypi/linux.git;name=machine;branch=rpi-6.6.y;protocol=https, attempting MIRRORS if available
ERROR: Fetcher failure: Fetch command
export PSEUDO_DISABLED=1;
export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus";
export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh";
export PATH="/home/ian/openbmc/scripts:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot/usr/bin/crossscripts:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot-native/usr/sbin:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot-native/usr/bin:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot-native/sbin:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot-native/bin:/home/ian/openbmc/poky/bitbake/bin:/home/ian/openbmc/build/tmp/hosttools";
export HOME="/home/ian"; LANG=C git -c gc.autoDetach=false -c core.pager=cat -c safe.bareRepository=all -c clone.defaultRemoteName=origin clone --bare --mirror https://github.com/raspberrypi/linux.git /home/ian/openbmc/build/downloads/git2/github.com.raspberrypi.linux.git --progress failed with exit code 128, see logfile for output

The logfile
--2025-02-19 16:47:47-- http://downloads.yoctoproject.org/mirror/sources/git2_github.com.raspberrypi.linux.git.tar.gz
Resolving downloads.yoctoproject.org (downloads.yoctoproject.org)... 37.27.135.249, 2a01:4f9:3070:3985::2
Connecting to downloads.yoctoproject.org (downloads.yoctoproject.org)|37.27.135.249|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-02-19 16:47:47 ERROR 404: Not Found.
--2025-02-19 16:47:48-- https://mirrors.kernel.org/yocto-sources/git2_github.com.raspberrypi.linux.git.tar.gz
Resolving mirrors.kernel.org (mirrors.kernel.org)... 139.178.88.99, 2604:1380:45e3:2400::1
Connecting to mirrors.kernel.org (mirrors.kernel.org)|139.178.88.99|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-02-19 16:47:49 ERROR 404: Not Found.

Asking for help.

kraj, 2025-02-20:
is this persistent ?

iankao0000, 2025-02-21:
I've tried serval times, all comes to the same ERRORs.

kraj, 2025-02-21:
On the build machine try cloning `git://github.com/raspberrypi/linux.git;` outside yocto fetching it. Perhaps in /tmp folder 

`git clone https://github.com/raspberrypi/linux.git`

does this work ?

iankao0000, 2025-02-21:
Yes, it does work.

kraj, 2025-02-21:
> Yes, it does work.

ok in that case, I think you might have to look at your project and see what could be firewalling bitbake from fetching it.  It works ok for me using latest master branch. ( I do not use openBMC setup though )