agherzan/meta-raspberrypi #1426: Core-image-x11 on CM5

sefram, 2025-02-25:
Hello,

I'm encountering an issue with the X11 environment on the Raspberry Pi Compute Module 5 (CM5) paired with the CM5IO board. After building and deploying a core-image-x11 using Yocto, the system boots to a black screen displaying only the mouse cursor; the Matchbox terminal does not launch.

Setup Details:

Yocto :

Build Configuration:
BB_VERSION           = "2.9.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi5"
DISTRO               = "poky"
DISTRO_VERSION       = "5.1.2"
TUNE_FEATURES        = "aarch64 crypto cortexa76"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "styhead:1cbcbab0555c8005b66dc31d2567bb9b3469d237"
meta-qt5             = "buildpath-fixes:5d61a3a0082ab5b9c30823bb6b4fc0302d78e9b7"
meta-oe              
meta-networking      
meta-python          = "styhead:6851cbf026088a13b2b45f684838d3dc4d7ccfdd"
meta-raspberrypi     = "master:923ac579c78ca42d425c303c34b65e199470ef67"
meta-python2         = "master:f02882e2aa9279ca7becca8d0cedbffe88b5a253"
meta-das1820         = "styhead:1cbcbab0555c8005b66dc31d2567bb9b3469d237"

local.conf includes:
MACHINE = "raspberrypi5"

Observations:
The same image functions correctly on a standard Raspberry Pi 5, with X11 and Matchbox terminal operating as expected.
On the CM5 + CM5IO setup:
X11 starts, showing only the mouse cursor on a black screen.
Matchbox terminal fails to launch.

sefram, 2025-02-26:
> i got same problem on a rasberry pi5 with x11 and matchbox manager. I have a mouse cursor on a black screen. Build Configuration: BB_VERSION = "2.9.1" BUILD_SYS = "x86_64-linux" NATIVELSBSTRING = "universal" TARGET_SYS = "aarch64-poky-linux" MACHINE = "raspberrypi5" DISTRO = "poky" DISTRO_VERSION = "5.1" TUNE_FEATURES = "aarch64 crypto cortexa76" TARGET_FPU = "" meta meta-poky meta-yocto-bsp = "master:ae8fe5b896a1f6bd0740d1fa6638b9d9377a8d28" meta-raspberrypi = "master:7494a378188f5d6330e038a5d6b9372b4aa232dd" meta-oe meta-python meta-multimedia meta-networking meta-gnome meta-filesystems meta-xfce = "master:0b83551ff2fefb043a5bd4a76a5c36cc36e2a2b4"

Helllo,
It's strange because for me, it's working on my Pi5 with core-image-x11. 
In addition, i just add that I try it on CM5lite without success.

Alanfromgard, 2025-03-01:
I bought a new RASPBERRY 5 with 4G.
And it is  working !
is there a revision problem ?
My previous non working rpi5 run with rasberry pi5 os , but not with yocto. 
A few days ago I update it with the last firmware, because it was not working.
My new one work with weston wayland image.

Where I can get the rev on the card. I don't see him ?

Alanfromgard, 2025-03-01:
My new card is rev 1.0. It is working with yocto.
My previous card is rev 1.1, and not working with yocto but with raspberry os.
Is it a rev problem ?

sefram, 2025-03-03:
Where is located REV on Pi5 and CM5 ?

sefram, 2025-03-03:
> Where is located REV on Pi5 and CM5 ?
I found it : 
My Pi5 is rev 1.0.
My CM5 is rev 1.0 (but there is no rev 1.1 on CM5)

sefram, 2025-03-03:
[Xorg.0_onPi5rev1.0.txt.log](https://github.com/user-attachments/files/19052057/Xorg.0_onPi5rev1.0.txt.log)

[Xorg.0_onCM5rev1.0.txt.log](https://github.com/user-attachments/files/19052059/Xorg.0_onCM5rev1.0.txt.log)

Attached, both Xorg.log on the 2 som. 
Only difference is dri1/2 maybe due to enumeration + glamor release.

sebjam22, 2025-04-08:
Dear all, 

I have the same issue with Yocto Kirkstone and Scarthgap.
But I have fixed it on my system (after long time of work). This issue is linked to the Mesa version.
For the Raspberry PI5, Glamor requires to have the version 7.1 of V3D.
So, I install Mesa 24.2.8 and now Xfce running correctly.

I share me recipe with you, keep in memory than I beginner and this recipe is not fully operational. For example, I use INSANE_SKIP to bypass the packaging.

So, I download the file mesa_24.2.8.orig.tar.xz on the site http://archive.raspberrypi.com/debian/pool/main/m/mesa/

Here, this is my recipe.
[mesa.zip](https://github.com/user-attachments/files/19654694/mesa.zip)

Here, this the patch.
[patch.zip](https://github.com/user-attachments/files/19654701/patch.zip)

Let's me know if this one working for you.

sefram, 2025-04-10:
> INSANE_SKIP

Hi @sebjam22 ,

I tried installing Mesa in my image. The build completes successfully. There's some improvement: I now get a working grafical terminal, but the FPS is very low ! 
It seems that hardware graphics acceleration isn't enabled.

sebjam22, 2025-04-10:
Hi @sefram ,

I think your setting is wrong.

In my config file (/root/config.txt):

        gpu_mem_256=256
 
        # Enable VC4 Graphics
        dtoverlay=vc4-kms-v3d-pi5
        max_framebuffers=2 
        disable_fw_kms_setup=1 
        arm_boost=1

In the folder : /etc/X11/xorg.conf.d, I have a file to force X11 to use modesetting named 99-vc4.conf included:
          Section "OutputClass"
           Identifier "vc4"
           MatchDriver "vc4"
           Driver "modesetting"
           Option "PrimaryGPU" "true"
          EndSection

Check the log of X11 (file : /var/log/Xorg.0.log) and you must have:
            [     6.829] (II) LoadModule: "glamoregl"
            [     6.830] (II) Loading /usr/lib/xorg/modules/libglamoregl.so
            [     6.837] (II) Module glamoregl: vendor="X.Org Foundation"
            [     6.837]    compiled for 1.21.1.14, module version = 1.0.1
            [     6.837]    ABI class: X.Org ANSI C Emulation, version 0.4
            [     6.876] (II) modeset(0): glamor X acceleration enabled on V3D 7.1.10.2
            [     6.876] (II) modeset(0): glamor initialized

If V3D 7.1.10.2 didn't appears you should be in software computation (rast...)

Also, you should have :
            [     7.351] (II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
            [     7.351] (==) modeset(0): DPMS enabled
            [     7.351] (WW) modeset(0): Option "PrimaryGPU" is not used
            [     7.352] (II) modeset(0): [DRI2] Setup complete
            [     7.352] (II) modeset(0): [DRI2]   DRI driver: vc4
            [     7.352] (II) modeset(0): [DRI2]   VDPAU driver: vc4
  
You can check than linux included the drivers (see below):

[image]

sefram, 2025-04-11:
Hi @sebjam22,

I check my config.txt --> everything looks fine (I've attached it).
I also looked into the Xorg.log, and indeed this line seems problematic:
`[     7.994] (II) modeset(0): glamor X acceleration enabled on softpipe`
I reviewed the kernel configuration (comparing with your option), and everything appears correct to me."

[Xorg.0.log](https://github.com/user-attachments/files/19701941/Xorg.0.log)

[config.txt](https://github.com/user-attachments/files/19701944/config.txt)

sebjam22, 2025-04-12:
Dear @sefram,

Yes, this is slow because the hardware accelaration is disable. The system works with software computation.

It seems than the frame buffer driver is missing on your system.
[     7.667] (II) LoadModule: "fbdev"
[     7.667] (WW) Warning, couldn't open module fbdev
[     7.667] (EE) Failed to load module "fbdev" (module does not exist, 0)

On my system,
[     6.667] (II) LoadModule: "fbdev"
[     6.667] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[     6.667] (II) Module fbdev: vendor="X.Org Foundation"
[     6.667]    compiled for 1.21.1.14, module version = 0.5.0
[     6.667]    Module class: X.Org Video Driver
[     6.667]    ABI class: X.Org Video Driver, version 25.2
[     6.667] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[     6.667] (II) FBDEV: driver for framebuffer: fbdev
[     6.700] (II) modeset(0): using drv /dev/dri/card0
[     6.700] (WW) Falling back to old probe method for fbdev
[     6.700] (II) Loading sub module "fbdevhw"
[     6.700] (II) LoadModule: "fbdevhw"
[     6.700] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[     6.701] (II) Module fbdevhw: vendor="X.Org Foundation"
[     6.701]    compiled for 1.21.1.14, module version = 0.0.2
[     6.701]    ABI class: X.Org Video Driver, version 25.2

So, I think than driver is missing in the linux kernel. However, I don't know which driver is exactly.

Here, my linux config.
[ConfigLinux.txt](https://github.com/user-attachments/files/19718304/ConfigLinux.txt)

Screen shot of the framebuffer section
[image]

And simple Framebuffer Support is activate
[image]

You are on the good way !

sebjam22, 2025-04-12:
Sorry, 
I forget to indicate this.

MACHINE_FEATURES should be include "vc4graphics"
MACHINE_FEATURES:append = " pci screen vc4graphics"

DISTRO_FEATURES should be include "opengl" and "x11"
DISTRO_FEATURES:append = " pci systemd usrmerge alsa dbus prootam opengl x11 vulkan wayland pam sysvinit"

sebjam22, 2025-04-14:
Hello,

Do you have included the DTS for the CM5 in the machine configuration file ?
Like that 
#@TYPE: Machine
#@NAME: RaspberryPi 5 Development Board (64bit)
#@DESCRIPTION: Machine configuration for the RaspberryPi 5 in 64 bits mode

require conf/machine/include/arm/armv8-2a/tune-cortexa76.inc
include conf/machine/include/rpi-base.inc

MACHINE_FEATURES += "pci"
MACHINE_EXTRA_RRECOMMENDS += "\
    linux-firmware-rpidistro-bcm43455 \
    bluez-firmware-rpidistro-bcm4345c0-hcd \
    linux-firmware-rpidistro-bcm43456 \
    bluez-firmware-rpidistro-bcm4345c5-hcd \
"

RPI_KERNEL_DEVICETREE = " \
    broadcom/bcm2712-rpi-5-b.dtb \
    broadcom/bcm2712d0-rpi-5-b.dtb \
    broadcom/bcm2712-rpi-cm5-cm5io.dtb \
    broadcom/bcm2712-rpi-500.dtb \
"

RPI_KERNEL_DEVICETREE_OVERLAYS += " \
    overlays/vc4-kms-v3d.dtbo \
    overlays/cma.dtbo \
    overlays/vc4-kms-v3d-pi5.dtbo \
    overlays/bcm2712d0.dtbo \
"

SDIMG_KERNELIMAGE ?= "kernel_2712.img"
SERIAL_CONSOLES ?= "115200;serial0"

VC4DTBO ?= "vc4-kms-v3d-pi5"

# When u-boot is enabled we need to use the "Image" format and the "booti"
# command to load the kernel
KERNEL_IMAGETYPE_UBOOT ?= "Image"
# "zImage" not supported on arm64 and ".gz" images not supported by bootloader yet
KERNEL_IMAGETYPE_DIRECT ?= "Image"
KERNEL_BOOTCMD ?= "booti"

#UBOOT_MACHINE = "rpi_arm64_config"

sefram, 2025-04-14:
@sebjam22 : I just added that in my local .conf : 

RPI_KERNEL_DEVICETREE = "bcm2712-rpi-cm5l-cm5io.dtb"

sefram, 2025-04-14:
Hi @sebjam22 ,

Can you confirm that you build an image for CM5 ?
in local.conf , you can find MACHINE ??= "raspberrypi5" isnt'it ?

sebjam22, 2025-04-14:
Hi @sefram,

Yes, I confirm building a image for CM5 (see picture)

[image]

Yes, my local.conf include MACHINE = "raspberrypi5". So, this machine is modified like post previuosly.

I think youd need to add overlays/cma.dtbo, overlays/vc4-kms-v3d-pi5.dtbo, overlays/bcm2712d0.dtbo overlay. You can check if these files are in the "boot" folder.

If you want, I can send you my build image but I don't know how.

sebjam22, 2025-04-15:
Hi @sefram 

On my PI5, I have the same issue than the CM5. Now, my build running correctly on both.

Perhaps this difference is liked to the processor revision, see next document page 4 (I have revision "0D0T" on my PI5 and CM5)

[PCN-Raspberry-Pi-5B-Rev-4.pdf](https://github.com/user-attachments/files/19753100/PCN-Raspberry-Pi-5B-Rev-4.pdf)

I use Yocto Scarthgap, Linux 6.6.78 and the raspberry-firmware is on date of 20230509.

sefram, 2025-04-16:
Hi @sebjam22,

You were right about the processor revision:

- My Pi5 is rev C1T
- And my CM5 modules are D0T

The issue is now resolved — X11 is running correctly with graphical acceleration!

To sum up:

- I needed to install Mesa 24.2.8, as you explained.
- I also added the following parameters to my local.conf:
`PACKAGECONFIG:append:pn-mesa = " gallium vc4 v3d gbm egl dri3"`
v3d was missing in this line. 
Since v3d wasn’t enabled, graphical acceleration wasn’t working properly before.

Huge thanks for your help, @sebjam22!

danielfdickinson, 2025-07-02:
Spoke a little too soon. If `udev` fails to start then the terminal correctly renders but input freezes. If `udev` starts properly then the black X screen with the not visible (but running if one checks from a serial console) matchbox-terminal occurs again.

crl-manoj, 2025-07-20:
> > Where do we get the libclc ?
> 
> It is in meta-clang layer (or in the master branch of oe-core, if you are using the bleeding edge version)

@OldManYellsAtCloud 
Thanks. I tried using the libclc from that layer now I'm getting all sorts of error while populating sdk. I am using Yocto scarthgap. Do you know what this error might be?

| Checking for function "pthread_setaffinity_np" with dependency threads: YES
| Run-time dependency expat found: YES 2.6.4
| Library m found: YES
| Run-time dependency libdrm_intel found: YES 2.4.120
| Run-time dependency libdrm found: YES 2.4.120
| Found CMake: NO
| Run-time dependency libudev found: NO (tried pkgconfig and cmake)
|
| ../mesa-24.3.1/meson.build:1811:2: ERROR: Problem encountered: llvmpipe requires LLVM and is enabled, but LLVM is disabled.

danielfdickinson, 2025-07-29:
@crl-manoj Did you use `CLANGSDK = "1"` as per the [meta-clang README](https://github.com/kraj/meta-clang/blob/master/README.md)?