agherzan/meta-raspberrypi #1427: Wifi connection problem

sapiippo, 2025-02-26:
I'm unable to get Wifi to connect on RPi4 when using styhead build.
Connection works correctly when using scarthgap build.

```
root@raspberrypi-armv8:~# wpa_supplicant -c /etc/wpa_supplicant.conf -i wlan0
Successfully initialized wpa_supplicant
[   55.047288] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
wlan0: Trying to associate with SSID 'wifi'
wlan0: Associated with xxxxx
wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan0: Authentication with xxxxx timed out.
wlan0: Added BSSID xxxxx into ignore list, ignoring for 10 seconds
nl80211: send_event_marker failed: Source based routing not supported
wlan0: CTRL-EVENT-DISCONNECTED bssid=xxxxx reason=3 locally_generated=1
wlan0: BSSID xxxxx ignore list count incremented to 2, ignoring for 10 seconds
wlan0: CTRL-EVENT-DSCP-POLICY clear_all
wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD

```

Tested with core-image-base using raspberrypi-armv8 MACHINE config.
poky styhead + latest meta-raspberrypi master

kraj, 2025-02-26:
what happens when you use MACHINE = "raspberrypi4-64"

sapiippo, 2025-02-27:
no change, still fails with the timed out message.

sapiippo, 2025-03-06:
Solution seems to same as used for RPi5 in  #1419 
Does some one know if the 'minimal' firmware is still needed for some rpi version or can we switch to always using the standard? @JPEWdev?

JPEWdev, 2025-03-06:
Based on [this post][1], it appears that the intention of the minimal firmware is that it uses less internal memory on the WiFi chip, which gives it more memory to track clients when in AP mode(?). I suspect it does this by sacrificing some standards comparability, and we would be better off defaulting to the standard firmware. Users can switch to minimal if they are wanting to e.g. host a web server on WiFi. This is also what Raspberry Pi OS does, with an `update-alternative` that allows the switch.

[1]: https://forums.raspberrypi.com/viewtopic.php?t=349490

omrisarig13, 2025-05-21:
Just to add more info from other boards: I have an RPi-CM4 board, and I'm also experiencing the exact same problem (able to connect to WiFi on Scarthgap, but not on Styhead), and the same solution (using the standard FW instead of the minimal) also fixes the issue for me.

MrTarantoga, 2025-08-11:
Same problem here with raspberrypi 3 64bit. Seems to be the [same error]( https://archlinuxarm.org/forum/viewtopic.php?t=17007) on arch an related to the version 2.11 of wpa_supplicant. Maybe related to removed patches as in arch?

MrTarantoga, 2025-08-26:
The solution for raspberrypi3-64 seems to add ```brcmfmac.feature_disable=0x02000``` to the ```cmdline.txt``` explained in https://github.com/RPi-Distro/firmware-nonfree/issues/34#issue-1523890517

Added a small bbappend in ```recipes-bsp/bootfiles/rpi-cmdline.bbappend``` with

```Python
CMDLINE_PATCH_WLAN_RPI3_ISSUE ?= "${@oe.utils.conditional("PATCH_RPI3_WLAN_ISSUE", "1", "brcmfmac.feature_disable=0x02000", "", d)}"
CMDLINE:prepend = " \
    ${CMDLINE_PATCH_WLAN_RPI3_ISSUE} \
```

To add this patch with an easy toggle in my configuration:

```
PATCH_RPI3_WLAN_ISSUE = "1"
```