agherzan/meta-raspberrypi #1431: Guidance for Secure Boot with Raspberry PI and this layer

BuriKizilkaya, 2025-03-06:
Does anyone have documentation on how to add the Secure Boot feature? 

Links
* [Raspberry-Pi-4-Boot-Security.pdf](https://pip.raspberrypi.com/categories/685-app-notes-guides-whitepapers/documents/RP-004651-WP/Raspberry-Pi-4-Boot-Security.pdf)
* [How to use[(https://pip.raspberrypi.com/categories/685-app-notes-guides-whitepapers/documents/RP-003466-WP/Boot-Security-Howto.pdf)

AshishShrikhande2408, 2025-03-09:
I am also looking for secure boot for PI5. Any steps would help a lot.

farias-automac, 2025-06-06:
> There exists a tool to provisioning with secure boot + full disk encryption:
> 
> https://github.com/raspberrypi/rpi-sb-provisioner
> 
> I've been using rpi-sb-provisioner. It works well.
> 
> It is built on the techniques here:
> 
> https://github.com/raspberrypi/rpi-eeprom/

Thanks! I'll check it out!