agherzan/meta-raspberrypi #1434: rpi5 configuration for GPIO UART

joaohf, 2025-03-14:
Hi,

For rpi5, I want to use GPIO (pints 14/15) UART. Does anyone know the correct configuration to be use with meta-raspberrypi ?

I've built a core-image-base that is working but as I don't have the pi debug UART cable/device, I would like to try the GPIO instead.

Thanks.

jackxl, 2025-03-18:
Have you tried 'enable_uart=1' ?

This stuff is configured using the config.txt and works the same as with raspbian.
I'd recommend checking the documentation for the rpi5.
https://www.raspberrypi.com/documentation/computers/config_txt.html#enable_uart

joaohf, 2025-03-19:
Hi,

I'm using the following configuration:

```
DISABLE_SPLASH = "1"

VIDEO_CAMERA = "0"

DISABLE_RPI_BOOT_LOGO = "1"

ENABLE_SPI_BUS = "1"

ENABLE_I2C = "1"

# Enable serial console on Raspberry PI systems
ENABLE_UART = "1"
```

I suppose 'ENABLE_UART' will make the right configuration at config.txt.

I'll also double check if my usb device is working fine.

joaohf, 2025-03-20:
Here is cmdline.txt and config.txt generated by meta-raspberrypi:

[cmdline.txt](https://github.com/user-attachments/files/19356074/cmdline.txt)

[config.txt](https://github.com/user-attachments/files/19356075/config.txt)

enable_uart is 1.

gizahNL, 2025-04-08:
If you want boot console output this isn't on pin14/15, it's on a separate header. It's possible with some tricks to get earlycon on pin 14/15, but you won't get U-Boot output (pin14/15 lives on the RP1 chip GPIO, which is connected via PCI-E, PCI-E support for Pi5 isn't in the U-Boot tree as of yet)

gizahNL, 2025-04-08:
See my comments here: https://github.com/agherzan/meta-raspberrypi/issues/1394 for some syntax regarding UART and earlycon (yours might differ since the CM5 ends up with slightly different addresses iirc).