agherzan/meta-raspberrypi #1453: synaptics-killswitch still required?

acostach, 2025-04-10:
Hi, quick question about https://github.com/agherzan/meta-raspberrypi/commit/7c31b9b4b595c3e0f932391e07a5258a9f966e4c :

Now that https://github.com/RPi-Distro/firmware-nonfree/pull/40 has been merged, is LICENSE_FLAGS = "synaptics-killswitch" kept for a different reason?

Thanks

agherzan, 2025-04-14:
I haven't followed up with that update but it looks good to me. @kappapiana what do you think?

ramisettyranganadh, 2025-05-17:
> Hi @agherzan , quick question about https://github.com/agherzan/meta-raspberrypi/commit/7c31b9b4b595c3e0f932391e07a5258a9f966e4c :
> 
> Now that https://github.com/RPi-Distro/firmware-nonfree/pull/40 has been merged, is LICENSE_FLAGS = "synaptics-killswitch" kept for a different reason?
> 
> Thanks

Works for me without it.