agherzan/meta-raspberrypi #1461: pi5 synaptics-killswitch is already in conf but cannot build core-image-base with linux-firmware-rpidistro

Proxy13, 2025-04-17:
hey there am i missing something or idk but seems i cant build core-image-base with linux-firmware-rpidistro 

[rubick@veargl31139-precision-3581 rpi-build]$ bitbake core-image-base
WARNING: Host distribution "arch" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |#################################################################################################| Time: 0:00:01
Loaded 4866 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'linux-firmware-rpidistro' (but /home/rubick/NAS/yoctoPi/poky/meta/recipes-core/images/core-image-base.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'linux-firmware-rpidistro' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['linux-firmware-rpidistro']
ERROR: Required build target 'core-image-base' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-base', 'linux-firmware-rpidistro']

my conf like this 

MACHINE = "raspberrypi5"
LICENSE_FLAGS_ACCEPTED = "synaptics-killswitch"

DISTRO_FEATURES:remove = "sysvinit"
DISTRO_FEATURES:append = " systemd usrmerge"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = ""
IMAGE_INSTALL:append = " systemd systemd-analyze systemd-serialgetty"

DISTRO_FEATURES:remove = "connman"
DISTRO_FEATURES:append = " networkmanager "
IMAGE_INSTALL:append = " networkmanager modemmanager"

IMAGE_INSTALL:append = " wifi-autoconnect"

IMAGE_INSTALL += "i2c-tools"
IMAGE_INSTALL:append = " nano net-tools tmux util-linux rsyslog kmod busybox openssl openssh openssh-scp openssh-sshd"

SYSTEMD_AUTO_ENABLE = "enable"
EXTRA_IMAGE_FEATURES += "ssh-server-openssh"
IMAGE_INSTALL:append = " packagegroup-core-ssh-openssh"

DISABLE_RPI_BOOT_LOGO = "1"
ENABLE_I2C = "1"
KERNEL_MODULE_AUTOLOAD:rpi += "i2c-dev i2c-bcm2708"

IMAGE_INSTALL += "linux-firmware-rpidistro"
LICENSE_FLAGS_ACCEPTED = "synaptics-killswitch"

Proxy13, 2025-04-17:
seems linux-firmware-rpidistro can be buildable standalone but can build with image-base

OldManYellsAtCloud, 2025-04-17:
The linux-firmware-rpidistro provides hardware specific runtime packages (like `linux-firmware-rpidistro-bcm43455`, which are already part of the respective machine config: https://github.com/agherzan/meta-raspberrypi/blob/master/conf/machine/raspberrypi5.conf#L9 - the needed firmware pieces should be installed by default. It should be enough to accept the license for this to work.

closed 2025-04-17 by Proxy13