agherzan/meta-raspberrypi #1478: Wrong ROOT_FS size

b4u-mw, 2025-06-05:
Hello,

I have set my rootfs size to a fixed size via

```
IMAGE_ROOTFS_SIZE = "524288"
IMAGE_OVERHEAD_FACTOR = "1.0"
IMAGE_ROOTFS_EXTRA_SPACE = "0 #"
```

(the comment in IMAGE_ROOTFS_EXTRA_SPACE is to ensure that nothing is added via "append").

I have printed the the calculated variables in poky/meta/classes-recipe/image.bbclass:get_rootfs_size() and there everythig is fine. But in the created WIC image, the rootfs images are ~650 MB.

I have no idea how to debug it further but I think it's within the wic file generation.

b4u-mw, 2025-06-06:
I saw that the filesystems (*.ext3, *.ext4) are also too big. I try to find out if this comes also from the raspberry layer or poky.

micw, 2025-06-06:
seems the easy way is just to set fixed-size in my wic file