agherzan/meta-raspberrypi #1519: graphics issues with different rpi5 revisions

sapiippo, 2025-09-01:
We have ran into graphical problems with Weston and Qt when running the same software on different HW revisions of RPi5.

For examples, with the two models below, first one is running fine while the second has display glitches on weston that makes it unusable.
```
Revision        : d04170
Model           : Raspberry Pi 5 Model B Rev 1.0

Revision        : e04171
Model           : Raspberry Pi 5 Model B Rev 1.1
```
This happens on walnascar when using core-image-weston recipe.

I suspect this is somehow related to mesa since with the latest poky master (Mesa 25.2.0) things work on all tested HW revisions.
Has anyone else seen this issue or know what's going on?

kraj, 2025-09-02:
Rev 1.1 boards need the D0 device-tree `bcm2712d0-rpi-5-b.dtb` and newer kernel/firmware. Perhaps that maybe helpful

sapiippo, 2025-09-02:
walnascar with mesa recipes from oe-core/master fixes this issue.
Seems that something in mesa has been fixed to work better on all RPi5 revision.