agherzan/meta-raspberrypi #1521: linux-raspberrypi-1_6.6.63+git-r0 do_fetch: Failed to fetch URL

martinboers, 2025-09-07:
I'm having a similar issue to #1315 - no matter what I try, the raspberrypi/linux source code cannot be fetched.

I'm attempting to do a simple Yocto build for rpi5:

```
$ git clone -b scarthgap https://git.yoctoproject.org/poky
$ cd poky
$ git clone -b scarthgap https://github.com/agherzan/meta-raspberrypi
$ source oe-init-build-env
$ bitbake-layers add-layer ../meta-raspberrypi
```

Additions to `local.conf`:

```
MACHINE = "raspberrypi5"
INIT_MANAGER = "systemd"
LICENSE_FLAGS_ACCEPTED = "synaptics-killswitch"
```

When I attempt to run all fetches:

```
$ bitbake core-image-minimal --runall=fetch
``` 

... all fetches complete successfully, except for the linux-raspberrypi recipe:

```
WARNING: linux-raspberrypi-1_6.6.63+git-r0 do_fetch: Failed to fetch URL ;protocol=https, attempting MIRRORS if available
ERROR: linux-raspberrypi-1_6.6.63+git-r0 do_fetch: Fetcher failure: Unable to find revision e442e5c1ab6bff5b5460b4fc949beb72aaf77970 in branch rpi-6.6.y even from upstream
ERROR: linux-raspberrypi-1_6.6.63+git-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/raspberrypi/linux.git;name=machine;branch=rpi-6.6.y;protocol=https')
ERROR: Logfile of failure stored in: /home/m/Projects/rpi5/poky/build/tmp/work/raspberrypi5-poky-linux/linux-raspberrypi/6.6.63+git/temp/log.do_fetch.6850
ERROR: Task (/home/m/rpi5/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_6.6.bb:do_fetch) failed with exit code '1'
```

The following commands both succeed, so there's no problem accessing the repository:

```
$ git clone -b rpi-6.6.y git@github.com:raspberrypi/linux.git

$ git clone -b rpi-6.6.y https://github.com/raspberrypi/linux.git
```

... and I have confirmed that branch `rpi-6.6.y` does contain commit e442e5c1ab6bff5b5460b4fc949beb72aaf77970

Since all other fetches succeed, I don't think it's a failure of the bitbake server (as suggested in #1315).

One thing I don't understand is this message in the log file:

```
fatal: repository 'https://github.com/git/raspberrypi/linux.git/' not found
``` 

This doesn't look like the correct URL - the correct URL should be either 'https://github.com/raspberrypi/linux.git/' or 'git@github.com:raspberrypi/linux.git', neither of which appear to have been tried.

I am prepared to accept that there is a configuration problem on my machine, but I can't figure out what it is or how to fix it. Any suggestions would be welcome.

OldManYellsAtCloud, 2025-09-08:
Another thing that I just noticed, is that your logs are mentioning `linux-raspberrypi-1` recipe, instead of `linux-raspberry-pi` which looks strange - you might want to run a search for such a recipe in your layers, and see what's up with it.

martinboers, 2025-09-21:
Thanks for the hints.

Based on this earlier report of a similar issue:

[Yocto Poky: Unable to fetch URL from any source.', 'git://github.com/raspberrypi/linux.git [SOLVED]](https://forums.raspberrypi.com/viewtopic.php?t=385507)

... I tried setting `BB_FETCH_TIMEOUT = "1200"` in `local.conf`, which did not help.

The `do_fetch task` appeared to be performing the "resolving deltas" stage exceptionally slowly, so eventually I tried `BB_FETCH_TIMEOUT = "6000"`, which worked. 

I guess I just have a really slow machine.

closed 2025-09-21 by martinboers